geopm/geopm #2892: geopmdpy.runtime does not restore controls after the run

bgeltz, 2023-03-14:
**Describe the bug**
I tried to run the static_agent.py tutorial Python agent and I expected controls to be restored to their initial value at the end of the run instead any requested control values persist after the run is complete.

**GEOPM version**
2c8cb6160

**Expected behavior**
```
$ geopmread CPU_POWER_LIMIT_CONTROL board 0
280

$ ./static_agent.py --geopm-report stress-report.report --geopm-initialize-control CPU_POWER_LIMIT_CONTROL=150 -- stress-ng --cpu 44 --taskset 0-43 --timeout 10

$ geopmread CPU_POWER_LIMIT_CONTROL board 0
280
```

**Actual behavior**
```
$ geopmread CPU_POWER_LIMIT_CONTROL board 0
280

$ ./static_agent.py --geopm-report stress-report.report --geopm-initialize-control CPU_POWER_LIMIT_CONTROL=150 -- stress-ng --cpu 44 --taskset 0-43 --timeout 10

$ geopmread CPU_POWER_LIMIT_CONTROL board 0
150
```

cmcantalupo, 2023-03-15:
I do not believe that this is unique to the python infrastructure.  The same code path exists for the GEOPM controller.

bgeltz, 2023-04-20:
I just verified that this is not a problem when using `geopmlaunch` as I was implementing #2918.  The reason is that `srun`, `mpiexec`, etc. do a `setsid` somewhere in their fork-exec logic.  This means that the GEOPM service tracks this new session as opposed to the one corresponding to your terminal in an interactive allocation.  As a result, any controls that are written by an Agent or similar are reverted when the job step ends.

PR #3113 (2023-09-15): Implement ServiceIOGroup save/restore
Fix bug in controller where controls accessed through the service are not restored. Add DBus API to restore controls (called by ServiceIOGroup) to allow controller to trigger controls to be restored.

- Fixes #2892
closed 2023-09-15 by cmcantalupo