geopm/geopm #3128: Plan integration test work to validate milestones and fill in backlog

cmcantalupo, 2023-09-20:
> estimate 1

lhlawson, 2023-09-22:
System Characterization Tests:
- [ ] ConstConfig IO can consume a dummy file and provide related signals via geopmread
- [ ] ParRes dgemm and nstream build using the ```./build.sh oneapi``` script & option
- [ ] gen_pals generates monitor scripts for pbs
- [ ] ParRes dgemm and nstream run successfully without GEOPM
- [ ] gen_pals generates gpu_frequency sweep pbs scripts for ParRes dgemm and nstream
- [ ] The GPU Frequency control requested is correct in the associated GEOPM reports
- [ ] The GPU CA Analysis script generates a const config file based on the GPU Frequency sweep reports
- [ ] Const Config IO can consume the GPU characterization file and provide the signals via GEOPM read
- [ ] AIB builds using the ./build.sh script 
- [ ] gen_pals generates a cpu_frequency sweep pbs script for AIB
- [ ] The CPU frequency control requested is correct in the associated GEOPM reports
- [ ] The CPU CA Analysis script generates a const config file based upon the CPU Frequency sweep reports
- [ ] Const Config IO can consume the CPU characterization file and provide the signals via GEOPM read
- [ ] Uncharacterized nodes provide a LEVELZERO signals for efficient frequency

lhlawson, 2023-09-26:
Board Power Limit Tests:
- [ ] BOARD_POWER_LIMIT related signals & controls exist.  Ex: BOARD_POWER_LIMIT, BOARD_POWER_TIME_WINDOW, BOARD_POWER, MSR::BOARD_ENERGY, MSR::PLATFORM_POWER_LIMIT:<PL1_POWER_LIMIT, PL1_TIME_WINDOW, PL1_LIMIT_ENABLE, LOCK>
- [ ] BOARD_POWER_LIMIT is adhered to as expected throughout high power job
- [ ] GEOPM Power Limit hook applies specified power limit at beginning of job
- [ ] GEOPM Power Limit hook removes specified power limit at end of job
- [ ] GEOPM Power Limit hook applies expected power limit in case of job crashing and subsequent job launch
- [ ] GEOPM Power Limit run vs workloads of interest
- [ ] GEOPM Power Limit sweep vs workloads of interest to determine performance-power tradeoff

dannosliwcd, 2023-09-26:
Non-uniform job power capping tests:
- [ ] Jobs execute immediately under no power or node contention
- [ ] Default job power limits are limited by the system budget
- [ ] Default job power caps are limited by the max node power
- [ ] Single-node jobs run concurrently when there is enough power
- [ ] Single-node jobs run in series when there is not enough power to run concurrently
- [ ] (if a backfill policy is enabled) Backfiller is not disrupted by automatic power cap selection
- [ ] (if a backfill policy is enabled) Backfiller respects system power limits

lhlawson, 2023-09-27:
GPU Activity Agent Tests:
- [ ] Agent functions without ConstConfigIO values
- [ ] Agent efficient frequency can be set by ConstConfigIO values
- [ ] Agent phi value modifies frequency range
- [ ] Agent responds to long periods of idle as expected 
- [ ] Agent phi sweep experiment runs to completion using ParRes DGEMM & NSTREAM
- [ ] Agent phi sweep experiments match expectations for ParRes DGEMM & NSTREAM
- [ ] Agent results match expectation with CORAL-2 workloads of interest