geopm/geopm #3255: make fails for the runtime build when using disable-geopmd-local without disable-docs

bgeltz, 2023-11-27:
**Describe the bug**
I tried to build the runtime on a system where only ```libgeopmd2``` and ```geopm-service-devel``` are installed and I expected it to complete successfully instead I encounter errors around the docs build.

**GEOPM version**
150c5ee74

**Expected behavior**
```
$ cd ${GEOPM_SOURCE}
$ ./autogen.sh
...
$ CC=icx CXX=icpx ./configure --disable-geopmd-local
...
$ make -j9
...
$ echo $?
0
```

**Actual behavior**
```
$ cd ${GEOPM_SOURCE}
$ ./autogen.sh
...
$ CC=icx CXX=icpx ./configure --disable-geopmd-local
...
$ make -j9
...
Warning, treated as error:
autodoc: failed to import module 'access' from module 'geopmdpy'; the following exception was raised:
No module named 'geopmdpy'
make[1]: *** [Makefile:10451: docs/build/man/.dirfile] Error 2
make[1]: Leaving directory '/mnt/scratch/bgeltz/geopm'
make: *** [Makefile:3684: all] Error 2
$ echo $?
2
```

**Additional context**
There are 2 issues here:
1. If you don't set your PYTHONPATH to point into the service directory, the base directory ```make docs``` target cannot find geopmdpy.  This can be worked around by issuing: ```PYTHONPATH=${GEOPM_SOURCE}/service:${PYTHONPATH} make -j9``` but that leads to the next problem:
2. By using the workaround above, you'll encounter this new error:
```
Warning, treated as error:
autodoc: failed to import module 'access' from module 'geopmdpy'; the following exception was raised:
cannot import name 'schemas'
make[1]: *** [Makefile:10451: docs/build/man/.dirfile] Error 2
make[1]: Leaving directory '/mnt/scratch/bgeltz/geopm'
make: *** [Makefile:3684: all] Error 2
```
This error is encountered because geopmdpy/schemas.py is a configure-time generated file.  You must run ```./autogen.sh``` and ```./configure``` in the service directory in order for this file to be available.

3. The easiest workaround is to use ```--disable-docs``` with the base directory build to avoid all of the above issues.  The successful set of commands is then:
```
$ cd ${GEOPM_SOURCE}
$ git clean -ffdx
$ ./autogen.sh
$ CC=icx CXX=icpx ./configure --disable-geopmd-local --disable-docs
$ make -j9
```

cmcantalupo, 2023-12-11:
I think adding either the `python3-geopmdpy` package or the `geopm-service` package should resolve this problem?  Is that not an option?

bgeltz, 2024-03-27:
Intention is to resolve this with the code reorganization PR #3230