geopm/geopm #3274: Check to see if cpufreq interface exists in sysfs, otherwise, throw at construction time with an appropriate message

cmcantalupo, 2023-12-05:


dannosliwcd, 2023-12-22:
I did a test by patching the iogroup use /sys/devices/system/cpu/cpufreqfake instead. Starting a debug run I see

```
Dec 22 09:18:59 mcfly1 geopmd[38407]: Warning: <geopm> Failed to load CPUFREQ IOGroup.  GEOPM may not work properly unless an alternate IOGroup plugin is loaded to provide signals/controls required by the Contr>
Dec 22 09:18:59 mcfly1 geopmd[38407]: The error was: <geopm> No such file or directory: SysfsIOGroup failed to open /sys/devices/system/cpu/cpufreqfake: at src/CpufreqSysfsDriver.cpp:33
```

A non-debug run emits no messages, but that's consistent with all other IOGroups

dannosliwcd, 2023-12-22:
Closing since it looks like our current implementation achieves this

closed 2023-12-22 by dannosliwcd