geopm/geopm #3292: Unable to start batch reads of LevelZeroIOGroup TIMESTAMP signals

dannosliwcd, 2023-12-18:
**Describe the bug**
I tried to add LEVELZERO::GPU_ENERGY_TIMESTAMP to batch request in `geopmsession` but it failed with an error that the signal can only used in batch requests.

**GEOPM version**
3.0.0

**Expected behavior**
GPU energy timestamps are printed after running `echo LEVELZERO::GPU_ENERGY_TIMESTAMP gpu 0 | geopmsession -t 1 -p 1`

**Actual behavior**
```
Error: geopm_pio_push_signal() failed: <geopm> Invalid argument: PlatformIOImp::push_signal(): unable to read signal name "LEVELZERO::GPU_ENERGY_TIMESTAMP" and domain type "8"
The following errors were observed:
<geopm> Runtime error: SDBus: Failed to call sd-bus function sd_bus_call_method(), error:-5 name: not.known.Error.RuntimeError: geopm_pio_read_signal() failed: <geopm> Invalid argument: PlatformIOImp::read_signal(): unable to read signal name "LEVELZERO::GPU_ENERGY_TIMESTAMP" and domain type "8"
The following errors were observed:
<geopm> Invalid argument: LevelZeroIOGroup::read_signal: TIMESTAMP Signals are for batch use only.: at src/LevelZeroIOGroup.cpp:1032
: at src/PlatformIO.cpp:778: at src/SDBus.cpp:32
: at src/PlatformIO.cpp:469
```

dannosliwcd, 2023-12-18:
It looks like this happens when PlatformIO checks whether a non-batch signal read succeeds as part of pushing a new signal.

dannosliwcd, 2024-08-23:
Note: This issue prevents users from computing average GPU power across windows different from the default 8-sample mean-power window.

cmcantalupo, 2025-02-18:
Although there are two other fixes for this problem:

https://github.com/cmcantalupo/geopm/commit/afb6a2753106e721008d25ae5b2ef43c1474d5e3
https://github.com/cmcantalupo/geopm/commit/76480610152e4ce6feee6f695eec6f26d3aaed33

The existing fix in dev:
https://github.com/geopm/geopm/commit/273f4faae7939061171bc21fc32149247d4581f0

works great:

```
$ echo "LEVELZERO::GPU_CORE_ENERGY_TIMESTAMP gpu_chip  0"|geopmsession -t 10 -p 1
"LEVELZERO::GPU_CORE_ENERGY_TIMESTAMP-gpu_chip-0"
1698.624597
1699.626067
1700.626067
1701.626047
1702.62605
1703.626052
1704.626063
1705.626054
1706.626055
1707.626088
1708.626079
```

closed 2025-02-18 by cmcantalupo