geopm/geopm #3516: Verify functionality of integration/apps infrastructure with out-of-tree builds

bgeltz, 2024-06-18:
> estimate 1
Parent [feature/story]: #XXXX

Due to <REPO_ROOT>/integration now having it's own separate build process, issues utilizing the apps infrastructure may be present.  Focusing on the apps that are needed for MS verification, check the following:

Done criteria:
- [ ] No issues exist for arithmetic_intensity or minife (or any apps that are used in WIP PRs)
- [ ] Experiments can still be run by hand (e.g. monitor, frequency sweeps)

Notes:
```
cd integration
CC=icx CXX=icpx FC=ifx F77=ifx F90=ifx ./configure --with-geopmd=${GEOPM_INSTALL} --with-geopm=${GEOPM_INSTALL} --disable-geopmd-local --disable-geopm-local --disable-fortran
```

dannosliwcd, 2024-06-20:
Testing with `3.1.1.dev9+gd81460568`

## Build
source ~/geopm/integration/config/build_env.sh
* `~/geopm/integration/apps/arithmetic_intensity $ ./build.sh` builds without errors
* `~/geopm/integration/apps/minife $ ./build.sh` builds without errors
* `CC=icx CXX=icpx FC=ifx F77=ifx F90=ifx ./configure --with-geopmd=${GEOPM_INSTALL} --with-geopm=${GEOPM_INSTALL} --disable-geopmd-local --disable-geopm-local --disable-fortran && make` builds without errors
* re-ran ./autogen.sh after git clean (since run_env.sh checks for VERSION)

## Run the experiment infrastructure directly
```
~/geopm/integration $ python3 -m experiment.run_experiment apps
amg
arithmetic_intensity
hpcg
hpl_mkl
hpl_netlib
minife
nasft
nekbone
pennant
qe
```

```
~/geopm/integration $ python3 -m experiment.run_experiment experiments
frequency_sweep
gpu_frequency_sweep
monitor
power_sweep
uncore_frequency_sweep
```

```
~/geopm/integration $ python3 -m experiment.run_experiment run monitor arithmetic_intensity --node-count $SLURM_NNODES
...
Run complete!
```

Verified that generated `arithmetic_intensity_sse_monitor_*.report` files contain each intensity region, and each of the hosts in the slurm reservation.

```
~/geopm/integration $ python3 -m experiment.run_experiment run monitor minife --node-count $SLURM_NNODES
...
Run complete!
```

Verified that generated `minife_monitor_*.report` files contain phases for matvec, dot, and waxpby, and each of the hosts in the slurm reservation.

## Run a test that invokes the experiment infrastructure
```
~/geopm/integration $ python3 -m test.test_cpu_characterization
Traceback (most recent call last):
  File "/usr/lib64/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib64/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/daniel/geopm/integration/test/test_cpu_characterization.py", line 34, in <module>
    class TestIntegration_cpu_characterization(unittest.TestCase):
  File "/home/daniel/geopm/integration/test/test_cpu_characterization.py", line 219, in TestIntegration_cpu_characterization
    @util.skip_unless_config_enable('beta')
  File "/home/daniel/geopm/integration/test/util.py", line 197, in skip_unless_config_enable
    if get_config_value('enable_{}'.format(feature)) == '1':
  File "/home/daniel/geopm/integration/test/util.py", line 156, in get_config_value
    with open(path) as config_file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/daniel/geopm/libgeopm/config.log'
```

Looking into workarounds now.

dannosliwcd, 2024-06-20:
`echo 'enable_beta=1' > /home/daniel/geopm/libgeopm/config.log` lets this test start running with the out-of-tree build.