geopm/geopm #3520: test_multi_app.sh rarely presents region exit without entry error

bgeltz, 2024-06-20:
**Describe the bug**
I tried to run test_multi_app.sh and I expected it to consistently pass instead it rarely fails.

**GEOPM version**
v3.1

**Expected behavior**
With test_mult_app.sh configured not to produce trace files, and with GEOPM_CTL_LOCAL present:
```
bgeltz@x4709c7s0b0n0:~/output> PATH=${HOME}/build/stress-ng/bin:${PATH} mpiexec -n 1 -ppn 1 ~/geopm/integration/test/test_multi_app.sh
/home/bgeltz/output
stress-ng: info:  [204962] setting to a 10 secs run per stressor
stress-ng: info:  [204962] dispatching hogs: 1 cpu
stress-ng: info:  [204962] skipped: 0
stress-ng: info:  [204962] passed: 1: cpu (1)
stress-ng: info:  [204962] failed: 0
stress-ng: info:  [204962] metrics untrustworthy: 0
stress-ng: info:  [204962] successful run completed in 10.00 secs
Beginning loop of 2 iterations.
Completed loop.

bgeltz@x4709c7s0b0n0:~/output> ls
temp_config.json  test_multi_app_report.yaml-x4709c7s0b0n0
```

**Actual behavior**
Exit before entry error, and no report files output:
```
bgeltz@x4709c7s0b0n0:~/output> PATH=${HOME}/build/stress-ng/bin:${PATH} mpiexec -n 1 -ppn 1 ~/geopm/integration/test/test_multi_app.sh
/home/bgeltz/output
stress-ng: info:  [204894] setting to a 10 secs run per stressor
stress-ng: info:  [204894] dispatching hogs: 1 cpu
Error: <geopm> Invalid argument: ValidateRecord::filter(): Region exit without entry Region=0x00000000: at src/ValidateRecord.cpp:70
Error: <geopm> Invalid argument: ValidateRecord::filter(): Region exit without entry Region=0x00000000: at src/ValidateRecord.cpp:70
stress-ng: info:  [204894] skipped: 0
stress-ng: info:  [204894] passed: 1: cpu (1)
stress-ng: info:  [204894] failed: 0
stress-ng: info:  [204894] metrics untrustworthy: 0
stress-ng: info:  [204894] successful run completed in 10.00 secs
Beginning loop of 2 iterations.
Completed loop.

bgeltz@x4709c7s0b0n0:~/output> ls
temp_config.json
```

**Additional context**
The above runs were done back-to-back with no changes in between.

bgeltz, 2024-06-21:
To reproduce this issue you need to have geopmctl sitting in the background when you try to run test_multi_app.sh.  This situation arose previously because I had somehow ended up in that state in my testing by trying a run, then issuing ^C to kill it because of an error, and re-running too quickly.

Repro steps:
```
bgeltz@x4709c7s0b0n0:~/output> export TEST_NAME=test_multi_app
bgeltz@x4709c7s0b0n0:~/output> export GEOPM_PROFILE=${TEST_NAME}
bgeltz@x4709c7s0b0n0:~/output> export GEOPM_PROGRAM_FILTER=geopmbench,stress-ng
bgeltz@x4709c7s0b0n0:~/output> export LD_PRELOAD=libgeopm.so.2.1.0
bgeltz@x4709c7s0b0n0:~/output> GEOPM_REPORT=${TEST_NAME}_report.yaml GEOPM_NUM_PROC=2 setsid geopmctl &

# Then before the Controller times out, issue:
bgeltz@x4709c7s0b0n0:~/output> PATH=${HOME}/build/stress-ng/bin:${PATH} mpiexec -n 1 -ppn 1 ~/geopm/integration/test/test_multi_app.sh
/home/bgeltz/output
stress-ng: info:  [187883] setting to a 10 secs run per stressor
stress-ng: info:  [187883] dispatching hogs: 1 cpu
stress-ng: info:  [187883] skipped: 0
stress-ng: info:  [187883] passed: 1: cpu (1)
stress-ng: info:  [187883] failed: 0
stress-ng: info:  [187883] metrics untrustworthy: 0
stress-ng: info:  [187883] successful run completed in 10.01 secs
Beginning loop of 2 iterations.
Error: <geopm> Invalid argument: ValidateRecord::filter(): Region exit without entry Region=0x00000000: at src/ValidateRecord.cpp:70
Error: <geopm> Invalid argument: ValidateRecord::filter(): Region exit without entry Region=0x00000000: at src/ValidateRecord.cpp:70
Completed loop.
bgeltz@x4709c7s0b0n0:~/output> Error: <geopm> Invalid argument: ValidateRecord::filter(): Region exited differs from last region entered Current region=0xf0e9be1c Received exit for=0xa12de8ee: at src/ValidateRecord.cpp:75
Error: <geopm> Invalid argument: ValidateRecord::filter(): Region exited differs from last region entered Current region=0xf0e9be1c Re
^C
```

dannosliwcd, 2024-07-17:
Alternative/additional expected behavior: when there's a left-over controller (e.g., from killing a monitored app but leaving the controller behind), we could improve error messages.

PR #3580 (2024-09-11): Add a profiler lock protecting the profile region name stack
- Fixes #3520
commit 0871ffe2 (2024-09-11): Add a profiler lock protecting the profile region name stack

- Fixes #3520
closed 2024-09-11 by cmcantalupo