geopm/geopm #3539: GEOPM reports no GPUs are present if the first usage of LevelZeroDevicePool is from the root user

dannosliwcd, 2024-07-24:
**Describe the bug**
I tried to login to a freshly updated node as root and I expected `geopmread -d` to show GPUs were present. Instead it showed 0 GPUs even though there were 6 on the system.

**GEOPM version**
GEOPM 3.1.0

**Expected behavior**
I expected `geopmread -d` to show GPUs were present. 

**Actual behavior**
Instead it showed 0 GPUs even though there were 6 on the system.

**Error messages produced**
No error messages were produced.

**Additional context**
The root cause ended up being that the first instantiation of LevelZeroDevicePool was in a context where ZES_ENABLE_SYSMAN=1 was not set. The documentation for **geopm_pio_levelzero(7)** specifies that the variable needs to be set. The geopm systemd service sets it automatically, but the first invocation of the device pool (which sets the topology cache) was from a root user, who does not go through the service.

Since the documentation specifies the environment variable is needed, and since we cannot guarantee that it is always set, we should emit a warning when that expectation is not met at the time our topology cache is being created.

cmcantalupo, 2024-10-12:
Calling setenv(2) should resolve the problem.  Previously we had issues when libmpi.so had already called zeInit prior to the construction of the geopm::LevelZero class.  Since GEOPM has been modified such that LevelZero is only loaded by root, and root should never be using MPI, this will not be a problem.

closed 2024-10-16 by cmcantalupo

cmcantalupo, 2024-10-31:
Work  around:

```
sudo systemctl restart geopm
```

PR #3616 (2024-10-16): Warn about required environment variables when using LevelZero
- Fixes #3539