geopm/geopm #3550: Some GPU signals sporadically show up in the report as NAN

bgeltz, 2024-08-08:
**Describe the bug**
I tried to use the GEOPM Runtime to monitor a simple microkernel that leverages the GPUs and I expected all the fields I added to the report via ```GEOPM_REPORT_SIGNALS``` to provide valid data instead several GPU signals reported ```nan```.

**GEOPM version**
v3.1

**Expected behavior**
The added report signals were:
```
GEOPM_REPORT_SIGNALS="TIME@board,BOARD_POWER@board,BOARD_POWER_LIMIT_CONTROL@board,CPU_POWER@package,CPU_FREQUENCY_STATUS@package,CPU_FREQUENCY_MAX_CONTROL@package,CPU_UNCORE_FREQUENCY_MAX_CONTROL@package,CPU_UNCORE_FREQUENCY_MIN_CONTROL@package,CPU_INSTRUCTIO
NS_RETIRED@package,CPU_PACKAGE_TEMPERATURE@package,CPU_UNCORE_FREQUENCY_STATUS@package,MSR::CPU_SCALABILITY_RATIO@package,MSR::QM_CTR_SCALED_RATE@package,DRAM_POWER@package,GPU_POWER@gpu,GPU_CORE_POWER@gpu_chip,GPU_CORE_FREQUENCY_STATUS@g
pu_chip,GPU_CORE_ACTIVITY@gpu_chip,GPU_UTILIZATION@gpu,GPU_UNCORE_ACTIVITY@gpu_chip,LEVELZERO::GPU_CORE_TEMPERATURE_MAXIMUM@gpu,DRM::THROTTLE_REASON_PL1@gpu,DRM::THROTTLE_REASON_PL2@gpu,DRM::THROTTLE_REASON_PL4@gpu,DRM::THROTTLE_REASON_PR
OCHOT@gpu,DRM::THROTTLE_REASON_RATL@gpu,DRM::THROTTLE_REASON_STATUS@gpu,DRM::THROTTLE_REASON_THERMAL@gpu,DRM::THROTTLE_REASON_VR_TDC@gpu"
```

```
GEOPM Version: 3.1.1.dev47+g6e066589d
Start Time: Thu Aug 08 15:03:17 2024
Profile: "sgemm_1_cpu_sweep_2"
Agent: monitor
Policy: {}

Hosts:
  node1:
    Application Totals:
...
      GPU_CORE_ACTIVITY@gpu_chip-0: 0.991445
      GPU_CORE_ACTIVITY@gpu_chip-1: 2.63989e+12
      GPU_CORE_ACTIVITY@gpu_chip-2: 0.991232
      GPU_CORE_ACTIVITY@gpu_chip-3: 0.993468
      GPU_CORE_ACTIVITY@gpu_chip-4: 0.995083
      GPU_CORE_ACTIVITY@gpu_chip-5: 0.993168
      GPU_CORE_ACTIVITY@gpu_chip-6: 0.994115
      GPU_CORE_ACTIVITY@gpu_chip-7: 0.988785
      GPU_CORE_ACTIVITY@gpu_chip-8: 0.994349
      GPU_CORE_ACTIVITY@gpu_chip-9: 0.99428
      GPU_CORE_ACTIVITY@gpu_chip-10: 0.990445
      GPU_CORE_ACTIVITY@gpu_chip-11: 1.30222e+11
      GPU_UTILIZATION@gpu-0: 0.992127
      GPU_UTILIZATION@gpu-1: 0.993384
      GPU_UTILIZATION@gpu-2: 0.995096
      GPU_UTILIZATION@gpu-3: 0.99245
      GPU_UTILIZATION@gpu-4: 0.995306
      GPU_UTILIZATION@gpu-5: 0.993061
      GPU_UNCORE_ACTIVITY@gpu_chip-0: 6.14837e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-1: 3.06554e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-2: 6.2169e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-3: 5.36591e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-4: 1.04582e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-5: 1.51411e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-6: 2.69546e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-7: 4.39557e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-8: 2.80187e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-9: 5.49371e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-10: 6.08198e-05
      GPU_UNCORE_ACTIVITY@gpu_chip-11: 2.81484e-05
...
```

**Actual behavior**
```
GEOPM Version: 3.1.1.dev47+g6e066589d
Start Time: Thu Aug 08 15:02:34 2024
Profile: "sgemm_1_cpu_sweep_1"
Agent: monitor
Policy: {}

Hosts:
  node1:
    Application Totals:
...
      GPU_CORE_ACTIVITY@gpu_chip-0: nan
      GPU_CORE_ACTIVITY@gpu_chip-1: nan
      GPU_CORE_ACTIVITY@gpu_chip-2: nan
      GPU_CORE_ACTIVITY@gpu_chip-3: nan
      GPU_CORE_ACTIVITY@gpu_chip-4: nan
      GPU_CORE_ACTIVITY@gpu_chip-5: nan
      GPU_CORE_ACTIVITY@gpu_chip-6: nan
      GPU_CORE_ACTIVITY@gpu_chip-7: nan
      GPU_CORE_ACTIVITY@gpu_chip-8: nan
      GPU_CORE_ACTIVITY@gpu_chip-9: nan
      GPU_CORE_ACTIVITY@gpu_chip-10: nan
      GPU_CORE_ACTIVITY@gpu_chip-11: nan
      GPU_UTILIZATION@gpu-0: nan
      GPU_UTILIZATION@gpu-1: nan
      GPU_UTILIZATION@gpu-2: nan
      GPU_UTILIZATION@gpu-3: nan
      GPU_UTILIZATION@gpu-4: nan
      GPU_UTILIZATION@gpu-5: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-0: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-1: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-2: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-3: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-4: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-5: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-6: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-7: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-8: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-9: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-10: nan
      GPU_UNCORE_ACTIVITY@gpu_chip-11: nan
...
```

**Error messages produced**
None

**Additional context**
These NaNs were produced on the first step of a frequency sweep.  The second step reported these fields properly.

Also, not sure what to make of:
```
GPU_CORE_ACTIVITY@gpu_chip-1: 2.63989e+12
...
GPU_CORE_ACTIVITY@gpu_chip-11: 1.30222e+11
```
These should always be < 1 AFAICT.

bgeltz, 2024-08-09:
I've been able to consistently reproduce this issue by issuing ```sudo systemctl restart geopm``` before attempting a run.  The first run after a restart always reproduces the issue.  The second run has no NaNs.

I was able to workaround this by reading any signal directly after service restart, but before starting the run:
```
sudo systemctl restart geopm
geopmread TIME board 0
<execute_run_script>
```

Utilizing the workaround forces the service to instantiate PlatformIO *before* the controller/runtime try to connect.  The reports come out fine after this until the service is restarted again.

asmaalrawi, 2024-08-28:
Occasionally seeing several hundred lines of nan during workload runs before GPU_ACTIVITY and GPU_UTILIZATION are populated. Values are sensible afterwards. This is while using geopmsession.

bgeltz, 2024-09-05:
Experiment with sample rate to see if that impacts how often this occurs.

bgeltz, 2024-09-06:
Re: chasing weird GPU_CORE_ACTIVITY values, add the component signals to the report to see if they are influencing the result (e.g. TIMESTAMP).

asmaalrawi, 2024-09-09:
GPU_CORE_ACTIVITY is being calculated as a derivative signal. In this case, when the GPU is not active, GPU_CORE_ACTIVE_TIME = 0 and GPU_CORE_ACTIVE_TIME_TIMESTAMP = 0. Derivative signal outputs "nan" cause 0/0. 

We confirmed, running SGEMM, that the timestamp is static when GPU is idle and as soon as the GPU has a load, the timestamp begins to update.

Possible solution:
* Accept that it's nan as there's no activity on the GPU
* Have the derivative output 0/0 = 0.  Problem: For other derivative signals, it can be more accurate to report nan. E.g. gaps in data are preferred when oversampling, rather than falsely getting 0s.
* Change derivative, allow each derivative signal to specify how to handle 0/0.

asmaalrawi, 2024-09-09:
Reporting issue: Should not be aggregating nans. 

Report is calling AvgAccumatorImp::update (and sum)--- change it to not update m_total and m_curr_total if there's a nan. Basically backfill nans with next sampled value.

bgeltz, 2024-09-09:
WIP for modifying the accumulator: https://github.com/bgeltz/geopm/commit/e9458d3e7bf1fc35ca26e84f8ee95389ea3c0728

Unsure if this is ultimately the proper approach here.

asmaalrawi, 2024-09-10:
Using GEOPM's TIME leads to large error. (values should be [0,1] and I'm seeing ~[0,12+]). 

Q: Is there something missing in this calculation? Using:

[ GPU_CORE_ACTIVE_TIME (t2) - GPU_CORE_ACTIVE_TIME {t1} ] / [t2 - t1]

[image]

PR #3594 (2024-09-20): Added nan handling for Accumulator sum and avg.
- Fixes #3592 
- Fixes #3550 

Sum and Average aggregators will skip NaN samples.
PR #3942 (2025-08-20): Assume full 64 bit width for activity counters
- Fixes #3550.
closed 2025-08-20 by bgeltz