geopm/geopm #3593: Ideal CPU/GPU maps associated with LevelZeroIOGroup do not match NUMA topology

dannosliwcd, 2024-09-19:
**Describe the bug**
I tried to view geopm reports on a GPU workload with annotated regions. Some GPU signals (e.g., GPU_POWER) were reporting `nan` in regions that were using those GPUs.

**GEOPM version**
v3.1.0

**Expected behavior**
All requested GPU signals are present in all regions that use GPU resources.

**Actual behavior**
GPU 2 had `nan` values in all signals in all regions.

**Additional context**
Expected topology can be viewed with `xpu-smi topology -m` or `grep '' /sys/class/drm/card*/local_cpulist`

GEOPM's GPU topology can be viewed with `sudo grep GPU /run/geopm/geopm-topo-cache`

The implementation of the LevelZeroIOGroup's topology is in `libgeopmd/src/LevelZeroGPUTopo.cpp`

The application was launched so that:
```
Localrank=0 : ZE Affinity mask = 0.0  CPU affinity list: 2,106
Localrank=1 : ZE Affinity mask = 0.1  CPU affinity list: 4,108
Localrank=2 : ZE Affinity mask = 1.0  CPU affinity list: 6,110
Localrank=3 : ZE Affinity mask = 1.1  CPU affinity list: 8,112
Localrank=4 : ZE Affinity mask = 2.0  CPU affinity list: 10,114
Localrank=5 : ZE Affinity mask = 2.1  CPU affinity list: 12,116
Localrank=6 : ZE Affinity mask = 3.0  CPU affinity list: 54,170
Localrank=7 : ZE Affinity mask = 3.1  CPU affinity list: 56,160
Localrank=8 : ZE Affinity mask = 4.0  CPU affinity list: 58,162
Localrank=9 : ZE Affinity mask = 4.1  CPU affinity list: 60,164
Localrank=10 : ZE Affinity mask = 5.0  CPU affinity list: 62,166
Localrank=11 : ZE Affinity mask = 5.1  CPU affinity list: 64,168
```

GEOPM's topology looked like:
```
GPU node0 CPU(s):   0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,204
  GPU chip0 CPU(s): 0,2,4,6,8,10,12,14,16,18,20,22,24,26,28,30,32,204
  GPU chip1 CPU(s): 1,3,5,7,9,11,13,15,17,19,21,23,25,27,29,31,33
GPU node1 CPU(s):   34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,205
  GPU chip2 CPU(s): 34,36,38,40,42,44,46,48,50,52,54,56,58,60,62,64,66,205
  GPU chip3 CPU(s): 35,37,39,41,43,45,47,49,51,53,55,57,59,61,63,65,67
GPU node2 CPU(s):   68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,206
  GPU chip4 CPU(s): 68,70,72,74,76,78,80,82,84,86,88,90,92,94,96,98,100,206
  GPU chip5 CPU(s): 69,71,73,75,77,79,81,83,85,87,89,91,93,95,97,99,101
GPU node3 CPU(s):   102,103,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,207
  GPU chip6 CPU(s): 102,104,106,108,110,112,114,116,118,120,122,124,126,128,130,132,134,207
  GPU chip7 CPU(s): 103,105,107,109,111,113,115,117,119,121,123,125,127,129,131,133,135
GPU node4 CPU(s):   136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155,156,157,158,159,160,161,162,163,164,165,166,167,168,169
  GPU chip8 CPU(s): 136,138,140,142,144,146,148,150,152,154,156,158,160,162,164,166,168
  GPU chip9 CPU(s): 137,139,141,143,145,147,149,151,153,155,157,159,161,163,165,167,169
GPU node5 CPU(s): 170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203
  GPU chip10 CPU(s): 170,172,174,176,178,180,182,184,186,188,190,192,194,196,198,200,202
  GPU chip11 CPU(s): 171,173,175,177,179,181,183,185,187,189,191,193,195,197,199,201,203
```

dannosliwcd, 2024-09-19:
https://spec.oneapi.io/level-zero/latest/core/api.html#_CPPv427zeDevicePciGetPropertiesExt18ze_device_handle_tP23ze_pci_ext_properties_t
provides the PCI address, which could be used to determine ideal CPU/GPU maps.

dannosliwcd, 2024-12-26:
WIP in #3569 appears to fix this issue. I spot-checked that change on a compute node that encounters this issue. Test automation still needs to be updated in the WIP.

Before:
```
GPU node0 CPU(s): 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,204
GPU node1 CPU(s): 34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,205
GPU node2 CPU(s): 68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,206
GPU node3 CPU(s): 102,103,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,207
GPU node4 CPU(s): 136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155,156,157,158,159,160,161,162,163,164,165,166,167,168,169
GPU node5 CPU(s): 170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203
GPU chip0 CPU(s): 0,2,4,6,8,10,12,14,16,18,20,22,24,26,28,30,32,204
GPU chip1 CPU(s): 1,3,5,7,9,11,13,15,17,19,21,23,25,27,29,31,33
GPU chip2 CPU(s): 34,36,38,40,42,44,46,48,50,52,54,56,58,60,62,64,66,205
GPU chip3 CPU(s): 35,37,39,41,43,45,47,49,51,53,55,57,59,61,63,65,67
GPU chip4 CPU(s): 68,70,72,74,76,78,80,82,84,86,88,90,92,94,96,98,100,206
GPU chip5 CPU(s): 69,71,73,75,77,79,81,83,85,87,89,91,93,95,97,99,101
GPU chip6 CPU(s): 102,104,106,108,110,112,114,116,118,120,122,124,126,128,130,132,134,207
GPU chip7 CPU(s): 103,105,107,109,111,113,115,117,119,121,123,125,127,129,131,133,135
GPU chip8 CPU(s): 136,138,140,142,144,146,148,150,152,154,156,158,160,162,164,166,168
GPU chip9 CPU(s): 137,139,141,143,145,147,149,151,153,155,157,159,161,163,165,167,169
GPU chip10 CPU(s): 170,172,174,176,178,180,182,184,186,188,190,192,194,196,198,200,202
GPU chip11 CPU(s): 171,173,175,177,179,181,183,185,187,189,191,193,195,197,199,201,203
```

After:
```
GPU node0 CPU(s): 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155
GPU node1 CPU(s): 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155
GPU node2 CPU(s): 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155
GPU node3 CPU(s): 52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,156,157,158,159,160,161,162,163,164,165,166,167,168,169,170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207
GPU node4 CPU(s): 52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,156,157,158,159,160,161,162,163,164,165,166,167,168,169,170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207
GPU node5 CPU(s): 52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,156,157,158,159,160,161,162,163,164,165,166,167,168,169,170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207
GPU chip0 CPU(s): 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155
GPU chip1 CPU(s): 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155
GPU chip2 CPU(s): 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155
[... cut ...] 0,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207
GPU chip11 CPU(s): 52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,156,157,158,159,160,161,162,163,164,165,166,167,168,169,170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207
```

closed 2025-01-04 by cmcantalupo

bgeltz, 2025-01-16:
Re-opened because the solution was reverted in #3663.

PR #3659 (2025-01-04): Map LevelZero GPUs to CPUs by PCI address
- Get GPU PCI addresses from LevelZero
- Use PCI addresses to ask Linux which CPUs are local to each GPU
- Resolves #3593