geopm/geopm #3642: geopmread/write requests sporadically fail

bgeltz, 2024-11-06:
**Describe the bug**
I tried to use geopmwrite to set a control and I expected it to work instead an error was presented.

**GEOPM version**
3.1

**Expected behavior**
```
[bgeltz@node1 ~]$ geopmwrite CPU_UNCORE_FREQUENCY_MIN_CONTROL board 0 2000000000
[bgeltz@node1 ~]$ echo $?
0
```

**Actual behavior**
```
[bgeltz@node1 ~]$ geopmwrite CPU_UNCORE_FREQUENCY_MIN_CONTROL board 0 2000000000
Error: cannot write control: <geopm> Invalid argument: PlatformIOImp::write_control(): control name "CPU_UNCORE_FREQUENCY_MIN_CONTROL" not found: at src/PlatformIO.cpp:828
```

**Error messages produced**
Upon inspection with a debug build, contention for dbus is observed:
```
node1 47232: Warning: <geopm> Failed to load SERVICE IOGroup.  GEOPM may not work properly unless an alternate IOGroup plugin is loaded to provide signals/controls required by the Controller and Agent.

node1 47232: The error was: <geopm> Runtime error: SDBus: Failed to call sd-bus function sd_bus_call_method(), error  :-110 name: org.freedesktop.DBus.Error.Timeout: Connection timed out: at src/SDBus.cpp:31

...

node1 47232: Error: cannot write control: <geopm> Invalid argument: PlatformIOImp::write_control(): control name "CPU_UNCORE_FREQUENCY_MIN_CONTROL" not found: at src/PlatformIO.cpp:828
```

**Additional context**
It seems we're overwhelming dbus with this request.  It's unclear how much other system traffic over dbus was occurring during the time of the failure.  Reproducing the failure is fairly sporadic.

cmcantalupo, 2024-11-06:
Work around posted here:
https://raw.githubusercontent.com/cmcantalupo/geopm/refs/heads/batch-write/geopmdpy/geopmbatchwrite.py

bgeltz, 2024-11-12:
The above workaround does not address the dbus calls that occur during the initialization of the ServiceIOGroup/PlatformIO.  Setting the ``SYSTEMD_BUS_TIMEOUT`` environment variable to a value of greater than 25 (the default) seems to allow the write requests to proceed as expected.
https://github.com/systemd/systemd/blob/main/docs/ENVIRONMENT.md?plain=1#L86

```
export SYSTEMD_BUS_TIMEOUT=600
```

bgeltz, 2024-12-03:
It seems that the dbus timeout was a red herring.  Looking through the syslogs on affected blades, I'm seeing messages like the following:
```
2024-11-26T12:25:27+00:00 node1 geopmd[147894]: The call io.github.geopm.PlatformGetUserAccess has failed with an exception:
2024-11-26T12:25:27+00:00 node1 geopmd[147894]: Traceback (most recent call last):
2024-11-26T12:25:27+00:00 node1 geopmd[147894]:   File "/usr/lib/python3.6/site-packages/geopmdpy/system_files.py", line 1232, in get_user_access
2024-11-26T12:25:27+00:00 node1 geopmd[147894]:     user_groups = self._get_user_groups(user)
2024-11-26T12:25:27+00:00 node1 geopmd[147894]:   File "/usr/lib/python3.6/site-packages/geopmdpy/system_files.py", line 1063, in _get_user_groups
2024-11-26T12:25:27+00:00 node1 geopmd[147894]:     return [grp.getgrgid(gid).gr_name for gid in all_gid]
2024-11-26T12:25:27+00:00 node1 geopmd[147894]:   File "/usr/lib/python3.6/site-packages/geopmdpy/system_files.py", line 1063, in <listcomp>
2024-11-26T12:25:27+00:00 node1 geopmd[147894]:     return [grp.getgrgid(gid).gr_name for gid in all_gid]
2024-11-26T12:25:27+00:00 node1 geopmd[147894]: KeyError: 'getgrgid(): gid not found: XXXX'
```

These groups are provided via LDAP in this case, so the current theory is that something is interfering with our ability to query for these groups when LDAP is in use.

The sequence as it's understood now:
1. User issues a geopmwrite.
2. Server-side geopmd begins servicing that request.  When it attempts to enumerate the groups associated with the user, the above error occurs.  Code: https://github.com/geopm/geopm/blob/release-v3.1/geopmdpy/geopmdpy/system_files.py#L1060-L1063
3. When the error occurs, geopmd crashes and is restarted by systemd.
4. The original geopmwrite request is never completed since geopmd dies, and eventually the client-side code hits a dbus timeout.

cmcantalupo, 2024-12-04:
Try to derive work around that does not require an updated version of the GEOPM service install.

PR #3649 (2025-01-02): Avoid using group database unless needed
- Fixes #3642
commit 8cef895c (2025-01-02): Avoid using group database unless needed

- Fixes #3642
closed 2025-01-02 by cmcantalupo